HALRobotics/Beta #2: Priority Loader Exception

sebandraos, 2017-09-17:
Occasionally, when loading grasshopper no login screen will be shown and instead a Priority Loader Error popup will appear with the message AsyncHelpers.RunTask... threw an exception. The error message indicates an aggregate exception but the nested elements are inaccessible.
This seems to be something to do with the licensing system but due to the sporadic nature of the bug I'm struggling to pinpoint the cause.

**Temporary solution:** 
_1. Close all instances of Rhino
2. Open the directory %LOCALAPPDATA%\HAL in a file explorer
3. Delete all *.lic files
4. Restart Rhino and Grasshopper and reenter your credentials_

This is still an issue with the below builds:
Installation Configuration:
HAL.ABB.1.0.17257.2-beta
HAL.KUKA.1.0.17256.2-beta
HAL.McNeel.V5.1.0.17260.3-beta
Licenses:
HAL.Core - 09/17/2017 22:54:22

sebandraos, 2017-09-18:
This should be fixed in HAL.McNeel.V5.1.0.17261.6 and above. Leaving open for the time being in case there are other factors at play

closed 2017-09-24 by sebandraos

yijiangh, 2018-06-18:
Hi @sebandraos , I installed HAL.McNeel.V5 1.1.15-beta, HAL.ABB 1.1.10-beta, and HAL.UR 1.1.9 beta on Rhino 5.0 in Win10. The installation was smooth but when I tried to launch grasshopper, GH got stuck at the following screen for 4 seconds:

[image]

And then there was an error message shown in the rhino command window:

`HAL for Grasshopper: The type initializer for 'HAL.Helpers.SerializationHelpers' threw an exception.
`
And there is no window poped up to enter credentials.

I am wondering if this problem is related to the issue reported above? I check the directory `%LOCALAPPDATA%\HAL` but found no license file there.

Thanks.

sebandraos, 2018-06-18:
@yijiangh I suspect this is more likely due to #22 . Please try uninstalling all other grasshopper plugins and seeing if that resolves the problem. Given the location of the error I suspect you have Newtonsoft.Json.dll in your Libraries directory which is of a different version to ours.

yijiangh, 2018-06-19:
You are right! Deleting other packages that have dependencies on Newtonsoft.Json.dll resolves the problem. Thank you!