HALRobotics/Beta #3: The plugin does not connect. And then "Priority: Loader" appears

K4rl33, 2017-09-18:
Dear developers,

As the other guy, I am also aexperiencing a  "Priority: Loader" error. 
But in my case I can't connect at the very start of the program, it keeps loading (I waited for 15 min) and then when I press the X button it gives me the error. 

I have no .lic files, but I tried reinstalling the program and the issue is still there.

Thank you in advance.

[image]

thibaultschwartz, 2017-09-18:
Hi K4rl33,
Could you please click on "copy as text" and post it here?

K4rl33, 2017-09-18:
I tried reinstalling the program, and unfortunately now I get another error (not the one of the upper window). 

I copy-paste the last one: 
```
Priority: Loader
{
  L'inizializzatore di tipo di 'HAL.McNeel.Grasshopper1.DocumentManager' ha generato un'eccezione.
  TypeInitializationException
}
```

sebandraos, 2017-09-19:
@K4rl33 can I ask you to please choose any installer/rhino instances you have open, go to %programdata%\HAL (you can copy and paste that in an explorer window) and delete everything that's in there. Then try reinstalling the extensions you want from the installer. That first exception is peculiar because that's not the version of that library we use. @thibaultschwartz can you check the app.configs for the McNeel packages?

sebandraos, 2017-09-19:
@K4rl33 I do, of course, mean close the installer/rhino instances

K4rl33, 2017-09-19:
Dear @sebandraos, I made your steps, but it still does not work:

[image]

It is in perpetual loading...

And if i close the window (after minutes):
```
Priority: Loader
{
  Non è stato possibile caricare il tipo 'System.IdentityModel.Tokens.Jwt.JwtSecurityTokenHandler' dall'assembly 'System.IdentityModel.Tokens.Jwt, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'.: System.TypeLoadException: Non è stato possibile caricare il tipo 'System.IdentityModel.Tokens.Jwt.JwtSecurityTokenHandler' dall'assembly 'System.IdentityModel.Tokens.Jwt, Version=4.0.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'.
   in HAL.Web.Licenser.Validate(String protectedText, IEnumerable`1& claims)
   in HAL.Web.Licenser.<RefreshAccessAsync>d__9.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.Web.Licenser.<RefreshLicensesAsync>d__13.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.McNeel.Grasshopper1.Client.Loader.#nERpOc'$-Qi2\\dx\*CGAj{rG#.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.Extensions.AsyncHelpers.<>c__DisplayClass0_0.<<RunSync>b__0>d.MoveNext()
   in HAL.Web.Licenser.Validate(String protectedText, IEnumerable`1& claims)
   in HAL.Web.Licenser.<RefreshAccessAsync>d__9.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.Web.Licenser.<RefreshLicensesAsync>d__13.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.McNeel.Grasshopper1.Client.Loader.#nERpOc'$-Qi2\\dx\*CGAj{rG#.MoveNext()
--- Fine traccia dello stack da posizione precedente dove è stata generata l'eccezione ---
   in System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   in System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   in HAL.Extensions.AsyncHelpers.<>c__DisplayClass0_0.<<RunSync>b__0>d.MoveNext()

  Exception
}

```

sebandraos, 2017-09-19:
This is very peculiar. I will take a deeper look this afternoon when I'm back in the office.

K4rl33, 2017-09-19:
@sebandraos I tried it on my Rhino WIP and it works!

K4rl33, 2017-09-19:
@sebandraos  I reinstalled Rhino and now HAL works well ;)

Thank you for the support!

sebandraos, 2017-09-19:
@K4rl33 that's great news. I will check if there are any packages that are tied down to English but I'm glad to hear you're up and running now. I will close this but feel free to reopen if the issue reappears.

closed 2017-09-19 by sebandraos