HALRobotics/Beta #6: HAL Components Failing to Compute upon Grasshopper Reload

bringley, 2017-10-01:
Unloading and loading Grasshopper in the same Rhino document session is causing multiple "archive is corrupt" Grasshopper IO errors and causing certain HAL Beta components to fail to compute.

This issue occurs when Grasshopper is unloaded and then reloaded during the same Rhino session.

This issue does not occur if Grasshopper is only loaded once per Rhino session.

I have attached the Grasshopper IO message log:
[HAL1SampleRouterToolpath_log.txt](https://github.com/HALRobotics/Beta/files/1347494/HAL1SampleRouterToolpath_log.txt)

Here is a screen capture of the HAL Create Tool component failing to compute:
[image]

This issue appears to stem from the fact that the HAL Grasshopper menu reloads and duplicates itself each time Grasshopper is unloaded and loaded again within a single Rhino session.
[image]

This issue occurs in the following Rhino version: Version 5 SR14 64-bit (5.14.522.8390, 05/22/2017)

And the following Grasshopper version: 0.9.0076

On the following operating system: Windows 10 Pro 1703 OS build 15063.608

Please advise and thank you for your time.

sebandraos, 2017-10-02:
Hi @bringley, this is a known issue caused by the fact that all of our components are generated at runtime. It is something that @thibaultschwartz will be looking into before the stable release. Thank you for the report.
**Temporary Workaround:**
Close and reopen Rhino entirely rather than unloading and loading Grasshopper

thibaultschwartz, 2017-10-05:
fixed in next release

closed 2017-10-05 by thibaultschwartz