HALRobotics/Beta #7: HAL Not Outputting RAPID Move Procedures

bringley, 2017-10-01:
HAL Beta is not correctly outputting ABB RAPID Move commands in my Grasshopper document.

The HAL Export component is merely outputting a line of RAPID that says "!Move" instead of the actual Move commands, despite being sent what I believe is a valid HAL Action Graph with 4 Move Actions.

[image]

Here is a close look at the Actions:
[image]

I tried this exact Grasshopper logic in the stable HAL (not the Beta) and it generated RAPID and ran on the robot perfectly.

I am attaching the Grasshopper document in quesiton wiht all Rhino references internalized:
[HAL1SampleRouterToolpath.zip](https://github.com/HALRobotics/Beta/files/1347511/HAL1SampleRouterToolpath.zip)

This issue occurs in the following Rhino version: Version 5 SR14 64-bit (5.14.522.8390, 05/22/2017)

And the following Grasshopper version: 0.9.0076

On the following operating system: Windows 10 Pro 1703 OS build 15063.608

Please advise and thank you for your time.

sebandraos, 2017-10-05:
Hi @bringley, We've just released a new version of the ABB extension which should have a few fixes in it. I am, however, going to leave this issue open because I wasn't able to reproduce your scenario.

bringley, 2017-10-09:
Hi @sebandraos I've updated to the new ABB extension for the HAL frameworks Beta and it appears to be an improvement, but I'm still getting some odd discrepancies between the RAPID generated in HAL 5.3 and the RAPID generated in the Beta. (Some of this may be a bug and some of this may simply be my misunderstanding of how to use the software.)

In the image below, I am showing the Beta RAPID output on the left and the 5.3 RAPID output on the right. This RAPID was generated by the same Grasshopper logic.

[image]

Main discrepancies:

- The Beta contains no declarations. I had thought that maybe the Alias input of components such as Speed Settings and Target from Frame would have served as its declaration as a named variable. Perhaps there is some other way to ensure that declarations are made in RAPID?

- The Beta only writes a MOD file, whereas 5.3 writes both a MOD and a PGF file within a task directory ("T_ROB1").

I am attaching a Grasshopper file that contains similar toolpathing routines in 5.3 and the Beta so you can try to reproduce.

[HAL-both-SampleRouterToolpath_002.zip](https://github.com/HALRobotics/Beta/files/1369771/HAL-both-SampleRouterToolpath_002.zip)

[image]

Please advise and thank you for your time.

sebandraos, 2017-10-09:
The underlying system for language export has been completely rewritten so it is likely that there will be slight differences from here on out. As for the bugs:
1. If you "shift + up" on the export component you will get a more complex version of the component which allows you to swap between inline or predeclared code. It defaults to inline to keep the files lighter. As most users generate code and never want to look at the stuff this seemed to make sense.
2. This is a known issue, strongly related to the uploading of code rather than just it's export. We have encountered an increasing number of cases where HAL code isn't the only code on a machine and we need to maintain, and modify, the existing pgf rather than ignoring the controller's content. We will be pushing it a solution for this along with the functional upload component.
Thanks for the continued help.

bringley, 2017-10-09:
Ah thanks for the tip regarding the predeclaration option - I'll test run this code against the 5.3 code at my controller this week to ensure I'm getting consistent motion from the Beta and do not run into any further RAPID issues.