HALRobotics/Beta #12: catalog supposed to display controllers displaying tools

ldemt, 2017-11-11:
Hello,

Here is my HAL 6 installation configuration : 

HAL.McNeel.V5.1.0.17278.2-beta
HAL.ABB.1.0.17278.2-beta

steps to reproduce the bug :
1) add robot, controller, and tool catalog components to the canvas
2) double-click tool, select cobra extruder, validate
3) double-click define controller constructor
bug appears : catalog displays tools and not controllers

sebandraos, 2017-11-11:
Thanks for the report @ldemt . We have noted this in our internal bug systems and will be working on a fix. Our internal testing shows that this issue exists only when two different catalog-based components are instantiated without an item being selected. For the time being , therefore, a temporary work around is to replace the component showing the wrong catalog and immediately use it for a new item. We will, however, find the source and patch it asap

sebandraos, 2017-11-29:
@ldemt Please note we do not currently support Rhino 6, any correct functionality is purely coincidental.

sebandraos, 2018-05-18:
This should be fixed in the latest updates. Please feel free to reopen if this is not the case.
P.S. Rhino 6 should be compatible now

closed 2018-05-18 by sebandraos