HALRobotics/Beta #13: robtarget names are missing in RAPID exported code when using export component in predeclaration mode

ldemt, 2017-11-12:
Hello,

Here is my installation configuration : 

Installation Configuration:
HAL.McNeel.V5.1.0.17278.2-beta
HAL.ABB.1.0.17278.2-beta

I'm exporting a single module (using the export component in predeclaration mode) for a simple ABB IRB120 setup in Robotstudio. The toolpath is a planar freeform curve discretized into several targets using "From Curve (Target)" component.

Here is the bug :

in the outputted .mod file the robtargets variable names are missing. They look like this : 

!robtarget
CONST robtarget :=[[441.465748,63.6,202.6],[0,0,1,0],[0,0,0,0],extjoint0];
CONST robtarget :=[[435.578421,34.013291,202.6],[0,0,1,0],[0,0,0,0],extjoint1];
...

but are supposed to look like this : 

!robtarget
CONST robtarget Curve_0_Start :=[[441.465748,63.6,202.6],[0,0,1,0],[0,0,0,0],extjoint0];
CONST robtarget Curve_0_End :=[[435.578421,34.013291,202.6],[0,0,1,0],[0,0,0,0],extjoint1];
...

In PROC part of the code, MoveL instructions are correctly written with good robtarget naming. As a workaround, it is possible to check this part of code and then add robtarget variable names by hand to the robtarget declarations. the .gh with internalized geometry, outputted .mod file are in the .zip attached

[1711-12-HAL6ExportBugAttachment.zip](https://github.com/HALRobotics/Beta/files/1464534/1711-12-HAL6ExportBugAttachment.zip)

sebandraos, 2017-11-12:
Thanks for reporting this @ldemt. We have found this in our own internal testing and have a fix ready for distribution. Unfortunately we have a number of changes that this fix will come with that aren't quite ready but we will endeavour to get this out to you and everyone by the end of the week.

sebandraos, 2018-05-18:
This should be fixed in the latest updates. Please feel free to reopen if this is not the case.

closed 2018-05-18 by sebandraos