HALRobotics/Beta #15: HAL Target component appears in grasshopper component dropdown menu when typing "mesh"

ldemt, 2017-11-15:
Hello,

Installation Configuration:
HAL.McNeel.V5.1.0.17278.2-beta
HAL.ABB.1.0.17278.2-beta

Here is the bug : when I type "mesh", the HAL target component appears first

[image]

Also when I type "move" i get the HAL Move component first. I use more often the vanilla grasshopper Move component so this is not comfortable imho.

Thank you

thibaultschwartz, 2017-11-16:
Hi Idemt,
This is due to a bad component suggestion sorting algorithm in Grasshopper, which does not prioritize native components. I could add some settings to entirely disable the registration of HAL content in the search menu, or add the presence of a prefix (i.e: HAL Move) to decrease the score of HAL components. I am adding this on the todo list.
Best

thibaultschwartz, 2017-11-17:
fixed in next release

closed 2017-11-17 by thibaultschwartz