HALRobotics/Beta #18: The component failed to compute! BUG REPORT

youcall911, 2017-11-29:
i wanna select ur5 but it says: 1. The component failed to compute! 找不到方法:“Rhino.Geometry.Mesh Rhino.Geometry.Mesh.CreateFromBox(Rhino.Geometry.Box, Int32, Int32, Int32)”。﻿ 

rsvp

sebandraos, 2017-11-29:
@youcall911 To be able to trck down this bug it is important that you tell us which build configuration you are currently using (you can see that in the About section of the HAL menu in Grasshopper), which version of Rhino and Grasshopper you are using, and how exactly you are going about selecting the UR5.

youcall911, 2017-11-29:
rhino version 5 (5.1.30103.145, 2013/1/3)
grasshopper build 0.9.0076

Installation Configuration:
HAL.McNeel.V5.1.0.17278.2-beta
HAL.UniversalRobots.1.0.17278.2-beta

Licenses:
HAL.Core - 11/29/2017 12:14:23

i just follow the youtube video and double click "robot" and search ur5, then jump out the error

youcall911, 2017-11-29:
@sebandraos

sebandraos, 2017-11-29:
@youcall911 Please ensure your Rhino 5 version is up to date. 5.1 is a few years old now, you should be on 5.12+ to ensure compatibility.