HALRobotics/Beta #19: UR Robots

sandarc, 2018-02-27:
Default UR Robots' (3, 5 and 10) starting position leads to collisions and joint violations when simulated/executed. Additionally, custom tool doesn't seem to rendered at the tool flange. 

GH version: 0.9.0076
Rhino version:  5 SR14 64-bit, 5.14.522.8390
OS: Windows 10 Pro

Installation Configuration:
HAL.McNeel.V5.1.0.17278.2-beta
HAL.UniversalRobots.1.0.17278.2-beta

Thanks
[image]

sebandraos, 2018-03-01:
Thanks for the heads-up. This seems to be an issue with UR robot presets missing a correctly implemented flange.
We are working on a substantial update at the moment which force us to update the presets and fix this. I can't promise any deadlines but we are actively working on lining up different bits of code to release ASAP.

thibaultschwartz, 2018-05-18:
fixed in next update

closed 2018-05-18 by thibaultschwartz