HALRobotics/Beta #21: Menus déroulant des composant robot/controller/Tool vides

carneaup, 2018-05-18:
Les menus déroulants pour choir le robot, le controller et l'outil ne contiennent aucun élément. J'utilise Rhino 5 sous windows 10. 
Ca fonctionne par contre sur mon autre ordi sous rhino 6.
[image]

sebandraos, 2018-05-18:
Are you connected to the internet on this computer? The catalogues are all hosted online with only previously used elements stored locally

carneaup, 2018-05-18:
Yes I am.

sebandraos, 2018-05-18:
Most likely is that your firewall is blocking Rhino's access to the internet. If that isn't the case please ensure your HAL installation is up to date and finally let us know which other Grasshopper plugins you have installed on your Rhino 5 PC compared to your Rhino 6 PC.

carneaup, 2018-05-19:
I tried to deactivate the Firewall completely but nothing changed. I also tried to disconnect Internet to see if I had an error message but it's still the same. 
On my other computer, it works both with Rhino 5 and Rhino 6.
On both computer I have the following additional plug-ins installed: HAL (last version), Karamba, Kangaroo2, Weaverbird, K2Eng and Anemone plus the ones by default. 
On the computer where HAL isn't working properly, I also have the followings: Fabtools, PanelingTools, Kangaroo, Lunchbox, Extra, gHowl and Droid.

sebandraos, 2018-05-19:
We have added a new feature to the latest version (available now) which will indicate if the catalogues can't connect to the servers. If that isn't visible, could you try temporarily uninstalling all other extensions and seeing if that fixes the problem? If so, we can work out with which extension there is a conflict and try and sort it out.

carneaup, 2018-05-19:
It's working now. There is conflict with the Newtonsoft.json.dll file. It works until I put this file in the gh libraires folder. I think I used it with the "heron" plug-in for GH but since I've been uploading and deleting many gh plu-ins, I'm not sure... Anyway, when I have this dll in the gh library folder, I have the same result as in the beginning without any additional comment (and I did the update which I assume is the new feature you mentioned).
[Newtonsoft.Json.zip](https://github.com/HALRobotics/Beta/files/2019834/Newtonsoft.Json.zip)

thibaultschwartz, 2018-05-20:
Thanks for the feedback, we were expecting this sort of dependency hell with other GH plugins, so we now have a  concrete case to test it. We will need a bit of time to figure how to get around this issue. The discussion on conflicting dependencies will continue on #22.

closed 2018-05-20 by thibaultschwartz