HALRobotics/Beta #23: Trouble with HAL robotics framework installer

bringley, 2018-05-23:
When attempting to update HAL.McNeel.V5, the installer gets hung up on component 9 (it doesn't matter whether or not I also try to update ABB/UR packages - it still gets stuck on component 9).

[image]

Things I've tried that haven't yet solved the issue:
*Restarting Rhino/Gh.
*Attempting the update using the HAL installer executable
*Uninstalling and reinstalling the HAL installer executable
*Attempting the update using the HAL menu within Grasshopper and then waiting for the installer to auto-launch after closing Rhino
*Logging out and back into HAL

Please advise and thank you for your time.

sebandraos, 2018-05-23:
@bringley if you can afford to not update until tomorrow I will push out a revised installer that has an external logging system so we can find out what the cause is. Unfortunately because our updates are incremental and modular, component 9 isn't a constant we can decipher. It would also be useful if you could zip C:\ProgramData\HAL and upload it so we can try and repro the issue on our systems.
If you are in need of an update now, however, please delete C:\ProgramData\HAL and reinstall from the installer (the installer itself won't be uninstalled by that deletion).
Again, if you can afford to hold out until tomorrow it would be great to get a read of the root cause of the issue as it's exceptionally difficult to reproduce this kind of bug.
Thanks in advance.

sebandraos, 2018-05-23:
If you can zip up the folder I mentioned, it would be great to have a repro at our disposal too.

sebandraos, 2018-05-24:
CI build of the updated installer is underway and should be live in 5 minutes. If you then try and perform the update again you should find a file "Installation.log" in C:\ProgramData\HAL. That should help us track down the problem.

sebandraos, 2018-05-24:
Did your installer update automatically before the latest installation attempt? You should have seen it restart.
I can see all sorts of legacy libraries in your zipped files which we should have cleaned up. They may well be causing the issues. I suggest deleting the folder and using the installer once again. That should create a fresh install and it should work properly.
You were a very early adopter (and we greatly appreciate that and your continued reports) but things have changed so drastically since the first few releases if the beta and it's probably better to start fresh at this stage.
If the installer works fine on a clean install I suggest we close this and reopen if any issues arise from here on out.

bringley, 2018-05-25:
Yes the installer did update. However, you were right, everything worked correctly by deleting ProgramData/HAL and installing the files from scratch. Thank you!

closed 2018-05-25 by bringley