HALRobotics/Beta #24: Display Problem of selection bar in controller or robot selection windows

nicolasDucou, 2018-05-25:
Hal configuration :
- Hal.McNeel.V5 version : 1.1.15-beta
- HAL.ABB version : 1.1.10-beta
 
I am using Rhinoceros 5.0 with Grasshopper version : 0.9.76.0, running on a virtual machine using VirtualBox with windows 10 lite.  

2. Action : I just select controller component and try to select one.

When i try to select a controller, double-clicking on the respected component, the list of available appears, but the blue bar permitting to validate the choice does not. Then I can select a controller but I cannot validate the choice.. That prevent me from using the software..
There is the same problem with the robot menu.
[BugDisplay.zip](https://github.com/HALRobotics/Beta/files/2039330/BugDisplay.zip)

sebandraos, 2018-05-25:
My guess is that this is a resolution issue rather than anything else, can you see the bottom of the window? Could you upload a screenshot of what you can see?

nicolasDucou, 2018-05-25:
No I think that I can't see the bottom of the window..

[image]

nicolasDucou, 2018-05-27:
I thank you for your answer.

Here is the screenshot of what I can see.
I effectively cannot see the bottom of the window..

sebandraos, 2018-05-27:
We had made an assumption about the minimum window size. We are working on a more adaptive solution and will update here once it's live.

nicolasDucou, 2018-05-28:
Indeed, increasing the resolution solved the problem!

Thank you for your help!

thibaultschwartz, 2018-05-29:
Low-resolution compatibility & re-sizable catalog windows are included in the next update.

closed 2018-05-29 by thibaultschwartz