HALRobotics/Beta #25: Inconsistent Mechanism Display

bringley, 2018-05-25:
Hello, it would seem that not all of the robot mechanisms are displaying properly out of the Robot component. For example, the UR5 and UR10 display correctly but the UR3 does not.

Also, this is more of a preference than a bug report, it would be nice if mechanisms displayed properly in a data param OR if the HAL params allowed for local display setting overrides (icon vs text).

Thank you!

bringley, 2018-05-25:
I looked into this further - the UR3 appears once it is fed a joint rotation list for the Position[] input. The UR5 and UR10 appear at their default positions without a Position[] input.

sebandraos, 2018-05-25:
I can't repro this. What version directory do you see under "%LocalAppData%\HAL\Mechanisms\f4f9dc46-7a7f-4c4c-834e-9e0a15d9e6b2"? It should be 0.18126.2, if it isn't please, once again, delete the folder mentioned and try again. Everything is supposed to auto-update if you are online but it is possible that there is a bug in the updater.

bringley, 2018-05-25:
Yes that is the version I see. Oddly enough, I see some UR3 files in this Mechanisms directory - should those be here? Is the UR3 mechanism data not getting serialized the same way as the other UR arms? See image:
[image]

sebandraos, 2018-05-25:
That all looks right. What do you mean by "not getting serialized the same way as the other UR arms"? If you find the UR5/10s in the Mechanisms folder they should look very similar. We use incomprehensible GUIDs as the identifiers of all content to enable name changes further down the line.
Could you please zip and send me the 0.18126.2 folder so I can see if I can repro? 
Is the UR3 working properly other than the initial display?

sebandraos, 2018-05-28:
The files I have on my PC are identical (down to their SHA-256 sums) so I'm not sure what could be causing this issue. I replaced my local version with your files and it worked exactly as expected. I can't think of any other differences between the loading of the UR3 vs UR5/10 other than these files so I must confess I'm a little stumped. I suggest, as before, deleting the UR3 folder and downloading a fresh copy through the catalogue. That shouldn't make any difference as the files seem to be ok but it might trigger some sort of refresh.
If the issue persists then I'll try and think of a way to dig deeper.

bringley, 2018-05-28:
Ok thanks for reviewing. I'll keep an eye on it to see if the issue persists and is consistent.

closed 2018-05-28 by bringley