HALRobotics/Beta #27: I have got an empty list in the robot component

madil4, 2018-06-03:
I have followed the installation procedure precisely but I have got an empty list when I doubled clicked the Robot component.

I'm running windows 10, and Rhino 5 640-bit.

[image]

sebandraos, 2018-06-03:
Possible duplicate of #21 and therefore #22 . Can you please try deactivating all other grasshopper plugins and seeing if that works? Please also let us know which versions of the HAL Robotics Framework components you have installed. You can see this in the HAL->About menu in grasshopper.

madil4, 2018-06-03:
It works now after removing Lunchbox plugin,  I'm wondering if the robot ABB IRB 6650 will be supported soon?

thibaultschwartz, 2018-06-04:
Yes the IRB 6xxx series presets will be added very soon. I close this issue and add a reference to #22 . Thanks for reporting this bug.

closed 2018-06-04 by thibaultschwartz