HALRobotics/Beta #30: Tutorial for configure HAL to work with external axis?

yijiangh, 2018-06-19:
Hi, I am wondering if we can have a tutorial on how to configure HAL to work with a 6-axis robot + linear track (prismatic joint)  setting? I am using HAL.McNeel 1.1.16-beta + HAL.ABB 1.1.11-beta.

With the linear track's geometry at hand, I was thinking about maybe this can be achieved by using the `create joint` and `assemble` component. A simple demo file could be extremely helpful. Many thanks in advance.

sebandraos, 2018-07-03:
External axes are not yet supported within the Framework. We are actively developing this feature and will be releasing a demo file alongside the feature release. I can't give any guarantees of timings at the moment but we should have it included in the beta in the next couple of months.

sebandraos, 2020-08-10:
External axes are now supported. There are a few tutorials [here](https://docs.hal-robotics.com/) that should help, most of which have demo files.

closed 2020-08-10 by sebandraos