HALRobotics/Beta #31: Notification about speed limit unclear

carneaup, 2018-06-22:
Hello,

You'll find attached a zip file with a tool path where I play with the speed of the robot. The file is working and the procedure is solving without error but there is a notification (INFO) about the speed that cannot be reached which is unclear. This message appears also if i don't give any speed setting to the robot.

The same message also appeared on a different file that we actually used on the robot without any problem (it appeared after I updated HAL last time).

Best regards,
Paul
[Speed limit unclear.zip](https://github.com/HALRobotics/Beta/files/2127197/Speed.limit.unclear.zip)

sebandraos, 2018-06-22:
Could you please explicit what is unclear about the message? I appreciate the second sentence is cut short but the former is complete and implies that we don't believe you can get to the defined speed due to some mechanical or trajectory constraint. It is purely informational so shouldn't have any negative impact on the solving or execution.

carneaup, 2018-06-26:
My concern is that this message appears now on all the files I'm working on
and since the second sentence is cut short, I don't know what seems to be
the problem. In addition it doesn't seem to be affected by the speed
settings or acceleration settings. Although this doesn't affect the solving
of the procedure and the robot manage to execute the tool path correctly.

sebandraos, 2018-07-04:
We are continually rewriting our solvers and this message was based on a condition which should have been superseded. You shouldn't receive this error unnecessarily from today's update. Apologies for the confusion

sebandraos, 2018-07-04:
It seems I spoke too soon. Certain targets in your demo file are still invoking the message and it doesn't look like the joints are the limiting factor.

tristangobin, 2018-09-19:
Fixed in current version.

closed 2018-09-19 by tristangobin