HALRobotics/Beta #32: Custom simulation with wait time

romainduballet, 2018-06-26:
Hi,
I have a issue with a _Custom Simulation_ action. To reproduce the issue : 
Create a _Custom Simulation_, with following input : 
Alias = StartHWC
Expression = "StartHWC;" (in rapid code).
Simulation = a _Wait_ (HAL Component). 
The _wait_  component is : 
Alias = StartHWCSimulation
Duration = 1.0

After we combine this procedure with other motion action the simulation doesn't start. 
if we simply add the wait simulation instead, it works. 

Best regards,

sebandraos, 2018-06-27:
Thanks for pointing this out to us. This is an ambiguity with the current presentation and treatment of the "Simulation". We currently consider that a simulation should be run in parallel to the main execution thread. Obviously with a Wait this means that nothing visibly happens. We are looking into making this less ambiguous and allowing a simulation to block the current execution.

sebandraos, 2018-07-30:
This should be fixed in the update coming this evening. Please feel free to reopen if the update doesn't solve the issue

closed 2018-07-30 by sebandraos