HALRobotics/Beta #33: Joint space and cartesian space error

romainduballet, 2018-06-26:
Hello, 

I have had some positioning issue between joint space and cartesian space. On the file attached the robot is asked to reach the same target in the two different spaces. The actual position in cartesian space shows an error varying from 1mm (x,y axis) up to 3mm (z axis). I cannot really explain why so I figured that might be a bug.

Best regards
[Dub_jointtargeterror.zip](https://github.com/HALRobotics/Beta/files/2137484/Dub_jointtargeterror.zip)

Romain

sebandraos, 2018-06-26:
This is likely because there is a blend the first time you hit the target and a stop point the second. If I remove the blends and check the joint positions in both hits there is a difference in the configuration of joints 4 and 6 of ±0.000009 rad which cancel each other out. All other joints are identical to that level of precision. Let me know if this doesn't account for the error and we can reopen the issue.

closed 2018-06-26 by sebandraos

romainduballet, 2018-06-26:
aha you must be correct ! sorry about that