HALRobotics/Beta #34: Target number limitation

romainduballet, 2018-07-10:
Hello, 

On this file, it seems that the solving won't succeed if we augment the number of target per layer. More precisely, on the left there is a line division of 20 operated for the main path. 20 seems to be a maximum here and I am not sure why. It does not seem to depend neither of the overall width of the object, nor of the blend sizes. 

My computer has 16 Gb ram

Best regards 

Romain
[Dub_targetnumber_100718.zip](https://github.com/HALRobotics/Beta/files/2179564/Dub_targetnumber_100718.zip)

sebandraos, 2018-07-10:
Thanks for reporting. This isn't directly related to the number of targets but rather the resulting distance between them. Please see #35 for details. Following one or both workarounds there should solve your problem.

closed 2018-07-10 by sebandraos

romainduballet, 2018-07-10:
Hey Seb, 

I understand this is not the number itself. However i tried to put the blend ratio at zero and it can"t solve. I may not have correctly understood the #35

romainduballet, 2018-07-10:
Ok I get it, it worked with the blend orientation at zero (?)