HALRobotics/Beta #36: Controller instantiation failed No preset is selected.

lionpeloux, 2018-09-10:
Hello,

I get this error when I try to load one of the controller : 

`1. [Warning] Controller instantiation failed No preset is selected.`

Am I missing something ? Only the "generic controller" seems to load.
I'm using the beta from last friday.

[image]

tristangobin, 2018-09-10:
This error probable comes from a dll conflict. A newton soft json us used in hal and other plug-in. We use the last version wich can be not the case for other plug-in. To solve this try delete any .dll in gh component folder.

lionpeloux, 2018-09-10:
Hi tristan,

I've cleaned up my GH/Library directory but still got the same issue. I suppose you use the same sort of JSON process to load controllers and robots. In my case robot loading is ok.

sebandraos, 2018-09-11:
Please make sure you have the appropriate extension installed for the controller you are trying to load. I.e if you are trying to use the ABB IRC5 make sure that the ABB installer is ticked on in the installer.

lionpeloux, 2018-09-11:
Ok, this might be the problem. But since I installed the last beta I cannot see the extensions anymore. I tried to uninstall HAL  / restart / do a fresh install but still I cannot see the extension.

[image]

sebandraos, 2018-09-11:
That looks to be the issue then. Could you please zip C:\ProgramData\HAL and send me a link (wetransfer or similar might be appropriate) so I can examine why the extensions aren't displaying. I would then suggest you delete that same folder and try to reinstall. Hopefully the extensions should show up then.

lionpeloux, 2018-09-11:
Here is the folder : https://www.dropbox.com/s/jzdbneqz4u1v168/HAL.zip?dl=0 <https://www.dropbox.com/s/jzdbneqz4u1v168/HAL.zip?dl=0>

lionpeloux, 2018-09-11:
Ok, I first uninstall HAL, then delete the folder `C:\ProgramData\HAL` then reinstalled HAL and was able to select the extensions.

[image]

sebandraos, 2018-09-11:
Excellent, thank you. Please close this if the installation of the required extensions fixes the issue and I will take a look at your installation configuration to see if I can make it more robust.

closed 2018-09-11 by lionpeloux