HALRobotics/Beta #47: Controller Node throws error: 1. [Error] Line creation failed. Null or invalid argument: rotation.

yck011522, 2018-09-18:
### Prerequisites

* [ ] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

Controller Node throws error: 1. [Error] Line creation failed. Null or invalid argument: rotation.

### Steps to Reproduce

1. Follow Video Tutorial up to 6:45: https://www.youtube.com/watch?v=2jnhRoDefW0&index=5&list=PLJK1IixFmcZe2vmfNQD_y-yBYFuT3kXlb

**Expected behavior:** No error

**Actual behavior:** Controller Node throws error: 1. [Error] Line creation failed. Null or invalid argument: rotation.

**Reproduces how often:** Every time on load

### Versions

Windows 10 Enterprise
Rhino 5
HAL.McNeel.V5 1.2.6-beta
HAL.ABB 1.2.6-beta

### Additional Information

Curve is internalized in GH file for sharing
[unnamed.zip](https://github.com/HALRobotics/Beta/files/2393428/unnamed.zip)

tristangobin, 2018-09-18:
Hi yck011522 , 
Thank's for sharing this. However, I did not acheave reproduce the error. What robot type and controller type did you use?
Could you try this definition? =>
[yck011522-01.zip](https://github.com/HALRobotics/Beta/files/2393459/yck011522-01.zip)

yck011522, 2018-09-18:
Note that I also have Rhino 6 installed.

yck011522, 2018-09-18:
[image]

tristangobin, 2018-09-18:
What is your model unit in rhino? Could you, if it is not the case, open rhino 6 in mm and then open grasshopper and the definition? 
Thank's

yck011522, 2018-09-18:
Okay, got it. It works in Rhino 6. 
I thought this version of HAL is for Rhino V.5. as that was shown in the description of the HAL installer.  Or it is?

tristangobin, 2018-09-18:
Actually it is for both. Glad it now works. I close this issue. Thank's you for your report.

yck011522, 2018-09-18:
Sure, thanks. I'm happy working with Rhino 6. But you may want to investigate why it doesn't work for Rhino 5, for many people are still sticking with Rhino 5..

closed 2018-09-18 by tristangobin