HALRobotics/Beta #48: Problem with cluster: reference frames seem to be binded

lionpeloux, 2018-09-19:
Hello,

I have isolated this strange behavior. 

In the following MWE I am creating two `reference frames` from the `XY plane`. Each one is distinct as the input plane is rotated by a certain input angle. The name of the frame is the angle.

If I put the components in a cluster an reproduce the same example with two clusters, then O1 appears to always equals O2. If I change I1 the it will change both O1 and O2. It seems that the frame component inside the cluster always point to the same object ...

Is this the intended behavior ? If so, how can I make reusable clusters for tool path generation relative to a certain reference frame ?

[image]

[image]

# How to reproduce

In the **same with cluster area**:

1. Put 90 in I1 and observe that O1 becomes 90 and tooltip on R of Cluster 1 becomes also 90
2. Put 145 in I2 and observe that O2 becomes 145 and tooltip on R of Cluster 2 becomes also 145
3. Now put your cursor on R of Cluster 1 and see that tooltip is 145 but O1 is still 90
4. Wire again R of Cluster 1 into O1 and see that now O1 becomes 145 while I1 is still 90 ....

# Config

`HAL 1.2.6 beta`
`Rhino 6 SR8`
`MACOS 10.13.6` 
`VMWare Fusion 10.1.0` with `WIN7 PRO SP1`

[01_feuillure_debugref.gh.zip](https://github.com/HALRobotics/Beta/files/2396862/01_feuillure_debugref.gh.zip)

tristangobin, 2018-09-19:
Hi Lionel, 
Thank's for reporting this.
Because we are using our own internal system to generate components, cluster are not supported so far. I will keep this open until we find a proper solution. 
All the best,

lionpeloux, 2018-09-19:
Hi Tristan,

I would say this feature is really important if one want to make reusable and parametrized tool path logic. Is there any alternative approach you would recommend to achieve that at the moment ?

thibaultschwartz, 2018-09-25:
HAL components are maintaining object references, which means that as soon as they get in a cluster, you will only see the result of the last running cluster solution. GH clusters have no way to cope with object reference maintenance, as they simply are sub-documents with a bit of decoration for the inputs/outputs. They can only work with components handling value types, or creating new instances of reference types (which is the default behavior in both Rhino and GH, but not in HAL as for us there is only one part/robot/tool/etc. instance, shared by many components, and not necessarily in the order specified by the GH wires).

So to wrap up:
- you can't use HAL components in clusters as we pass objects through parameters via reference (we don't pass copies). We deal with it in regular GH documents via some hacks, as GH is not designed to process objects this way. Clusters are - by design - incompatible with object references.
- if you want to use GH clusters, you could enclose some logic creating GH data that you can then plug into HAL components outside the cluster
- tool-path generators should be compiled, as it's the only way to maintain compatibility between HAL client applications, but I'm afraid we're not providing a public API for this yet

closed 2018-09-25 by thibaultschwartz