HALRobotics/Beta #50: Rhino Crash after last update

nicolasDucou, 2018-10-03:
[RhinoCrashDump.zip](https://github.com/HALRobotics/Beta/files/2441176/RhinoCrashDump.zip)
[RhinoCrashDump.zip](https://github.com/HALRobotics/Beta/files/2441181/RhinoCrashDump.zip)

### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

[Description of the issue]

### Steps to Reproduce

1. [First Step]
Updating through Grasshopper
2. [Second Step]
close Rhino as asked
3. [and so on...]
Open Rhino 
4. the openning of grasshopper generates the error

**Expected behavior:** [What you expect to happen]
I would like to be able to use Grasshopper and HAL 

**Actual behavior:** [What actually happens]
Rhino crash

**Reproduces how often:** [What percentage of the time does it reproduce?]
anytime
### Versions

Up-to-date Rhino 5 
 (Version 5 SR14 64-bit
(5.14.522.8390, 05/22/2017)
Educational Lab License )

Last release of HAL
Running on Windows 7 Entreprise Service Pack 1

### Additional Information

Any additional information, configuration, files or data that might be necessary to reproduce the issue.
[RhinoCrashDump.zip](https://github.com/HALRobotics/Beta/files/2441194/RhinoCrashDump.zip)

sebandraos, 2018-10-03:
Hi @nicolasDucou . Can you please also upload the installer log and extension config files which you will find at:
"C:\ProgramData\HAL\Installation.log"
and
"C:\ProgramData\HAL\Extensions.config"
This will help us find the source of the issue. Once you have uploaded these files, I suggest deleting the entire "C:\ProgramData\HAL" directory and reinstalling with all instances of Rhino closed.

nicolasDucou, 2018-10-04:
Hi Sebastien, 

I am sorry, I already deleted and reinstalled the software and it works fine. Then, I can't provide you the log and extension config files.. But, I have a suggestion about what could happen. When I made the update, I had not the adminstrator rights. Then, during the update, it might not have installed all the files needed  (without telling me) which finally created the error... 

I thank you for your support.

closed 2018-10-05 by sebandraos