HALRobotics/Beta #51: Kinematic Settings : data conversion failed

lionpeloux, 2018-10-03:
Hello,

I would like to release some contraints on my targets. Namely, I want my targets to stick to a certain plane, but I would like to tell the solver It can freely rotates the targets in this plane.
 
For a kuka target defined by XYZABC, this means that the solver can freely choose the value of A (rotation around tool Z axis).

I tried to use the `Kinematics` component but the following error always shows up when I plug a robot into the Mechanism input.

Moreover, how would you define a constraint around an axis ?  The `Constraints` component does not accept vector/line as an input for the `RotationAxis` and I did not find anything to define this `RotationAxis`. Am I missing something ? Is this feature yet implemented ?

Many Thanks,

[image]

My config :
`HAL 1.2.6 beta`
`Rhino 6 SR8`
`MACOS 10.13.6` 
`VMWare Fusion 10.1.0` with `WIN7 PRO SP1`

sebandraos, 2018-11-28:
Unfortunately this isn't yet usable. The components shouldn't really be live except that they avoid having inputs on other components which nothing can output. The kinematics is pretty much functional but we haven't yet finalised how this will work with exporting code. I will leave this open and update when possible.

sebandraos, 2020-08-10:
This is now live. With respect to the constraints and the issue you have in your screenshot: If you hover over the input it's not asking for a vector, it's actually an enumeration, you need to right-click and pick and axis from the options available.

closed 2020-08-10 by sebandraos