HALRobotics/Beta #52: Malware engines detected in HAL Frameworks executable

bringley, 2018-10-17:
Hello - I can't run the latest HAL frameworks executable (https://halinstaller.blob.core.windows.net/windows/Setup.exe) as certain components are being detected as malware. You can review for yourself at https://www.virustotal.com/#/file/86b300d6610a4c625f642b0380f1868668279e378f3c7379cdc4dc8b82199dec/detectionlease.
[image]

sebandraos, 2018-10-18:
Hi @bringley I've just pushed a new version of the installer and implemented a hashing system to ensure what is available for download is exactly what we uploaded. [Virus total](https://www.virustotal.com/#/url/0c352817d6456d6f910078e12ff0301297f637122acc99305e60730990baddd3/detection) isn't raising any issues with the link now. I've updated one of our dependencies for the installer and I'm hoping that has solved the problem once and for all.
The new installer should have a SHA-256 hash of 76B828E4D76BB5219F6FB50378CF4B93DC619E75449EAEB916AB8A1215816834 (case insensitive) but may take a few minutes to be rechached.

Closing as I believe it's resolved but please reopen if there are further issues.

As a slight aside, I'm not sure where I'd lean on 3 dodgy vs 63 clean scores (which include all the engines I have used on personal and company machines). Is this a software system blocking you from downloading and/or installing, or is it a manual verification you're doing as a precautionary measure?

closed 2018-10-18 by sebandraos

bringley, 2018-10-18:
Thank you, Sebastian. I was able to run this successfully. It was my corporate antivirus software that blocked me from installing, so I used Virus Total to help identify the specific components that were causing the trouble. However, upon installation I'm only seeing 4 components.
[image]

I will follow up with infosec here to see if components are still being blocked.

sebandraos, 2018-10-18:
Once the installer is up and running nothing else should really be aware it's being installed (per the usual Windows definition). This looks like an installation error, I would try the same procedure as #23 and see if that helps

bringley, 2018-10-18:
Ha - my own post! Yes that worked, thank you for everything.