HALRobotics/Beta #53: Mesh Extraction of HAL Object Types

bringley, 2018-10-21:
Is there currently a method for extracting mesh (or other standard Grasshopper geometry) objects from HAL objects such as a robot param? See image below comparing HAL Frameworks with a similar workflow in HAL.

[image]

Thank you,
Brian

sebandraos, 2018-10-23:
Hi @bringley . The elements of a robot are "a little" more complex than they were in HAL5 so geometry extraction is a slightly more convoluted affair. Can you give us a bit more information about your requirements for this? e.g. Is this for creating images? Are you doing calculations on the meshes that are being passed back into some algorithm? Do you want to run some custom collision analysis?

bringley, 2018-10-23:
Understood. Mostly it's for the creation of images, although yes things like returning geometry to another algorithm or custom collision work would potentially be useful in the future, too. I also like the joint rotation circles that display with the arm - I could see those being very useful for diagramming, or for creating joint traces in addition to TCP traces.

lionpeloux, 2018-10-24:
In the same idea, It would be very useful to be able to define some additional geometry rigidly attached to one part of the arm that will be "transported" during robot motion.

For instance, I have an algorithm that can simulate elastic rods. Lets say I want to simulate the geometry of a flexible hose attached between to parts of the arm. This way I can have at each simulation step the anchoring positions of my hose to simulate its position and do some verifications / analysis on it.

bringley, 2018-10-24:
That's a great point @lionpeloux. I have to admit that I don't quite understand how to attach parts to non-flange locations on an arm - in the GIF below I try to extract various parts of the arm but the only connection option is the default flange. Is there some means of defining a custom Connection?

[image]

I think HAL has historically played quite well with Grasshopper but I'd like to see more inputs capable of accepting default geometry object types and not have my hands tied by HAL object types, although once I learn how to define those object types with geometry it may no longer be an issue.

sebandraos, 2018-10-24:
There are all sorts of other considerations for extracting data at every simulation step (performance impacts, component child expiry etc.). But for rigid connections, you will find **Connection** under **Create Joint** and any **Parts** that aren't part of the mechanism can be passed to the **Environment** param of **Execute** to update their positions. 
We need to work on some clear documentation for all of this and, as you will no doubt notice very quickly, there are quirks to the system including immediate transformation of your part to its connected location.
Hopefully that helps you both continue for now. We will discuss the data extraction options internally bearing this conversation in mind and suggest a few possible implementations.

bringley, 2018-10-24:
Wonderful - thank you, @sebandraos !

tristangobin, 2023-08-09:
This is now available in 1.4-beta and next versions.

closed 2023-08-09 by tristangobin