HALRobotics/Beta #54: Traget Constraints and Mechanism Constraints

yck011522, 2018-11-18:
Hi, 

Great work so far.  I'm wondering how to use the TragetConstraints and MechanismConstraints from the Constraint component. 

Does it goes to a MotionSettings component?

thibaultschwartz, 2018-11-19:
Hi Victor,
We will add a tutorial on constraints at a later time, as it's still a bit experimental and the UI might still change quite a bit before it's stable.
Best,

ldemt, 2023-12-14:
Hello dear HAL Team,

Is there any updates on the Mechanism constraints component ? I'd like to use it for two reasons : 

- I have a robot mounted on a track and I'd like to lock the track movements for some motions,
- I'd like to limit the degree of rotation of some axes of my robot, because I have a robot dresspack and tool that might collide the robot's geometry if axis 5 turns too much for example,

Best,

sebandraos, 2023-12-15:
Hi @ldemt ,
The Mechanism Constraints should be able to do this in simulation but we are still having issues ensuring that the exported code makes the robots actually behave like the simulation. Essentially we are adding control over the robot using those Constraints that the real robots don't have built in. We are still developing tools that will automate some of this issue diagnosis and resolution.
At the moment, for the Track I would suggest using Grasshopper's inbuilt list management functionality to generate track positions for each of your robot's Targets, you'll get much better control that way, especially if you have explicit requirements like stopping the track at certain points.
And for axis 5 causing collisions, that's going to have to be handled with some Target modifiers. It might seem like more work but we've found we get more control by being explicit and having a rule-based solution. It would probably be true that in part we haven't prioritised the Constraints because we haven't actually found them to be very useful in most cases.