HALRobotics/Beta #58: UR Upload Procedure Incompatible with controller

sandarc, 2019-01-16:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

Upload Procedure used with HAL grasshopper sample file for UR. The command returns 1. [ERROR] Incompatible controller. The specified controller is not compatible with this feature.

### Steps to Reproduce

I Plugged in CB3 UR controller, solution per sample UR file (URDemo.gh), network id (IP), and Upload set to True. The command returns 1. [ERROR] Incompatible controller. The specified controller is not compatible with this feature.

### Versions

HAl version: 1.2.11-beta
OS: Windows 10 Pro

sebandraos, 2019-01-16:
Unfortunately this is an error in the demo file, we haven't completed the implementation of the UR upload. We will discuss this internally and see whether we can implement the UR upload in a reasonable time. If not, we'll correct the demo file.

sebandraos, 2019-03-23:
We have just released an update that adds upload functionality to the UR controller. This has been lightly tested (we don't have constant access to UR robots) so please let us know if there are any issues with the implementation.

closed 2019-03-23 by sebandraos