HALRobotics/Beta #61: Example request on "take/release" component

bigdeeer, 2019-04-23:
Hi Hal robotics!

I am using Hal robotics Grasshopper plugin to do some simulation, and I am trying to simulate something like below:
1:  Robot arm move to location A and pickup a bottle with the gripper.
2:  Robot arm move with the bottle.
3:  Robot arm move to location B and release the bottle.

I am planning to use one of the two ways below:
1:  After the procedure solver, retrieve the TCP plane of the final path and orient the bottle to the plane during the stimulation, so it look like the bottle is being grabbed with robot arm.
   Problem: I can not find a way to retrieve the TCP plane of the final path. 
2:  Use the "take/release" component to stimulate the grab and release action. 
  Problem 1: I am not sure this component is the right choice to do this.
  Problem 2: If this is the right component, I can not find help or example on how to use it. I checked Hal Youtube channel which is really helpful, but I can't get any help on how to use "take/release" component.

Would you please help me with the problem above?
Thank you very much!

Deer

tristangobin, 2019-04-23:
Hi Deer, 

Thank you for your message.
 
Please find attached  in a zip file a rhino file and a gh definition for basic take and release example.
I hope this will be enough for you to explore picking application. 
[ABBPickingExample.zip](https://github.com/HALRobotics/Beta/files/3106474/ABBPickingExample.zip)

Best,

bigdeeer, 2019-04-24:
> Hi Deer,
>
> Thank you for your message.
>
> Please find attached in a zip file a rhino file and a gh definition for basic take and release example.
> I hope this will be enough for you to explore picking [...]

Thank you very much!
This is really helpful.
It seemed that I did not change the "create tool" template to "gripper", and I kept getting "can not turn tool.tool into grasp.gripper" information.
Now it just works perfect.

Thank you again!

sebandraos, 2019-04-24:
Thanks @tristangobin 
I'm glad that's sorted @bigdeeer . It's good to know that the requirement for the tool to be a gripper isn't clear. We will ensure that is included in the videos/documentation when we produce them.

closed 2019-04-24 by sebandraos