HALRobotics/Beta #62: Can't load robots from the catalog

lionpeloux, 2019-05-02:
Hi,

My config :
HAL 1.2.19 beta
Rhino 6 SR8
MACOS 10.13.6
VMWare Fusion 10.1.0 with WIN7 PRO SP1

I've just upgraded to HAL 1.2.19 beta and I can't load any robots from the catalog.
Is this a known problem ?

[image]

Lionel

sebandraos, 2019-05-02:
Hi @lionpeloux , Can you please confirm that this issue is linked to COFF loading in the Grasshopper Developer Settings? i.e. If you turn COFF loading off, does the problem go away? This will help us narrow down the source of the problem.
Do you have any extensions installed?

sebandraos, 2019-05-02:
We have managed to repro this error. It appears we have a upgraded a dependency that is causing this issue. A new build is currently running and should be live in the next few minutes. Please check version 1.2.20 and let us know if the problem persists.

lionpeloux, 2019-05-03:
Hi seb,
I don't see the 1.2.20 update. Did it pass the build ?

sebandraos, 2019-05-03:
It did but I forgot to approve it, apologies. It should be live imminently

lionpeloux, 2019-05-03:
Ok, It's working now.
Thanks !

closed 2019-05-03 by sebandraos