HALRobotics/Beta #63: Take/Release simulation and code issues

anjakunic, 2019-05-11:
Dear HAL team,

I am having some troubles with take/release component in terms of simulation accuracy and code export:

1. When simulating the procedure, released parts end up randomly displaced compared to their corresponding release plane (see attached image).

2. I am using a Custom Action component to set the exact expression that my gripper is activated with. When not used with take/release component this action is properly exported in the script file and robot/gripper runs correctly. On the other side when I use it with take/release component, the exported script is not correct (see attached images).

I am attaching a gh file where both the issues can be observed.

Thank you very much in advance.
Anja

[image]
[image]
[image]
[image]

anjakunic, 2019-05-11:
gh definition:

[take-release trial01.zip](https://github.com/HALRobotics/Beta/files/3169480/take-release.trial01.zip)

tristangobin, 2019-05-13:
Hi Anja, 

I solve the simulation aspect of your issue. It was basically the accuracy of our solving that led to this trouble. It is because by default an interpolation (blend) is not 0. This leads to problems (like in the real world) when it comes to move part with accuracy. I started from your example file and did a pick and place example with proper blend and speed settings for each step of a standard pick and place routine. 
As you will see, this involves a bit more detailed movement since you need to describe each position precisely. 

For the code export problems, this is actually a bug that we have to solve on our side. 

I will agree that the pick and place could be a bit more simplified and we will work on that. 

Please find attached the gh file.

[image]

[take-release-advanced-example.zip](https://github.com/HALRobotics/Beta/files/3173185/take-release-advanced-example.zip)

anjakunic, 2019-05-13:
Hi @tristangobin,

Thank you for the fast feedback and solving the issue. The introduced blend and speed settings definitely provide a smoother and more refined toolpath, thanks for that too. 

For the code export bug, I can just use the algorithm version without the take/release component until it gets fixed.  

All the best,
Anja

sebandraos, 2019-06-03:
This is now fixed in our internal builds and will be included in the next update. Please reopen if the next update doesn't solve you issue.

closed 2019-06-03 by sebandraos