HALRobotics/Beta #65: Impossible to scroll in "custom controller"

6axis, 2019-06-27:
### Prerequisites

* [X ] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

The scroll bar in  "custom controller"  is locked after savage selection of another motion in the list. I'm not sure, but it lock when the line selection is done while simulation is running.

### Steps to Reproduce

1. Open cutom controller.
2. Run a target full simulation (more than 5000 targets). (time ratio at 10).
3. Scroll the list far down.
4. Random select a target.

**Expected behavior:** I'm expecting what it normaly do when it work : stop the simulation, update the position of the robot in the simulation, not lock the custom controller.

**Actual behavior:** Scroll list is locked. Closing and re-opening the "custom controller" is not doing any diffference. I have to quit Grasshopper, close Rhino and start from a freshly opened file.

**Reproduces how often:** For now, it seem like 80% chance that happend in my case.

### Versions

HAL
HAL.McNeel.V5 1.2.22-beta
HAL.ABB 1.2.21-beta
HAL.KUKA 1.2.13-beta
HAL.Staubli 1.2.6-beta
HAL.UniversalRobots 1.2.17-beta

Rhino 6 SR7

W10 PRO 64bits 17134.829

### Additional Information

In order to use my file, you have to load some Gcode in the corresponding cluster in grasshopper.

[Files.zip](https://github.com/HALRobotics/Beta/files/3333662/Files.zip)

tristangobin, 2019-06-27:
Hi 6axis, 
So far it's working as expected for me. 
The following reasons could be liked to your issues and it any case it will help you to work :
- SR7 => should be the last version of rhino 6, this cause geometry compatibility issue. I actually had trouble to use your file since the milling head brep was corrupted. I replaced it by a lighter version of the geometry. You will find that in the attached gh file.
- Geometry is heavy and cost a lot to display. Try to use lighter geometry, not something with more than 10000 faces. It helps.
- By hiding procedures display, you'll enhance the fluidity of the display. 
[Demonstrateur_Milling_Multi-Fixed.zip](https://github.com/HALRobotics/Beta/files/3333938/Demonstrateur_Milling_Multi-Fixed.zip)

6axis, 2019-06-27:
Hi tristangobin,  thank you for your answers. 

- Updating Rhino definitively improved the situation
- Ok, that's not a probleme, detailled model are not realy usefull in that case. Your right, that solved the problem too.
- Yeah ! It's not lagging anymore. 

Problem solved. Thank you.

Regards