HALRobotics/Beta #66: Attach Component errors when managing data trees.

sebastienmaitenaz, 2019-07-23:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

Hi, 
I'm on HAL.McNeel.V5 1.2.22-beta. 
I have a recurent mistake when I try to attach parent mechanisms ordered with a tree structure (let's say from {0;0} to {I;J}) with their corresponding tools or grippers (ordered the same way). 
It results in 15 mechanisms but only the last one is correct. I have no error message for the other 14 but they don't appear in the Rhino viewport and I can't solve procedures for them. 
However the parent mechanisms and the grippers are properly defined as everything works well if I explode both trees and attach parents with child separately it works.

Do you have any idea why this is happening ? 

Thank you,
Sébastien

sebandraos, 2019-07-23:
Hi @sebastienmaitenaz 
This is a known and very intentional restriction in that we don't handle trees in that component, or any that deal with mechanisms to avoid in-situ duplication of robots.
Out of curiosity, what is the use case where you're assigning 15 tools to 15 robots simultaneously?

sebastienmaitenaz, 2019-07-24:
Thank you @sebandraos
I have a gantry with 5 cranes on it to manipulate long objects. On each of the 5 cranes I have 3 mechanisms in order to manipulate several of this long objects at the same time. 
I am not sure to perfectly understand your answer as I am able to manipulate trees with other components dealing with mechanisms (Assemble component for example).