HALRobotics/Beta #67: Combined procedures solving issue

sebastienmaitenaz, 2019-08-23:
Hi,

I have an issue when combining procedures for a single prismatic joint mechanism.
When I'm assigning it a Move procedure alone, the solving succeeds. However, for some reason when I'm combining the Move procedure with other actions (Wait - Move - Wait for example), the solving is cancelled.  

[image]

Thank you,
Sébastien

sebandraos, 2019-08-23:
Hi @sebastienmaitenaz , 
Is there any chance you have a minimal grasshopper file you can share that reproduces this issue. We have a large number of demonstration files which handle various types of actions being merged so this should be a supported scenario.
Thanks

sebastienmaitenaz, 2019-08-23:
@sebandraos, here are a Rhino and a Gh demo files. 
Thank's !

[One_DOFs(R&P)_Demo_File.zip](https://github.com/HALRobotics/Beta/files/3534030/One_DOFs.R.P._Demo_File.zip)

sebastienmaitenaz, 2019-08-26:
@sebandraos, it actually seems combining procedures (includin a move action) works as long as the first action is Move. 
So a workaround is to set as first input a Move action with the initial position of the mechanism as a target, and then add other actions. But that's not a very convenient.

sebandraos, 2020-08-10:
This should be fixed in the latest update but please reopen if need be.

closed 2020-08-10 by sebandraos