HALRobotics/Beta #69: Grasshopper Loading Errors at Start

sebandraos, 2020-01-22:
### Description

When opening Grasshopper a series of error messages appear in a pop-up often listing `TargetInvocationException`s and/or `FileNotFoundException`s.

### Cause

This is almost always due to a corrupted installation but it is sporadic and therefore hard to pinpoint.

### Help to Fix

Please zip `C:\ProgramData\HAL` and upload it as a comment as this contains everything we need to examine the installation failure. Please also include the OS and what version of the OS you're running, a copy of the errors, and which installations and versions that you have installed. The latter can be found via the installer.

### Workaround

Once you have zipped the file and uploaded it so we can try and find the root cause of the issue you can delete `C:\ProgramData\HAL` and reinstall the Framework. Please make sure that Windows is up to date.

julienenpc, 2020-01-22:
Hello,
I Couldn't upload the HAL folder zipped (too big) so here it is 
https://filesender.renater.fr/?s=download&token=59c5c694-0c9b-a8cb-b057-2556af8c678f
I have windows 10 version 1607.
I use 1.2.33 for Mcneel and 1.2.32 for ABB

Here the copy of the errors 
Object: GHC_Attach (level 1)
{
  Une exception a été levée par la cible d'un appel.
  TargetInvocationException
}

Object: GHC_Attach (level 2)
{
  Impossible de charger le fichier ou l'assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' ou une de ses dépendances. Le fichier spécifié est introuvable.
  FileNotFoundException
}

Object: GHC_Solver (level 1)
{
  Une exception a été levée par la cible d'un appel.
  TargetInvocationException
}

Object: GHC_Solver (level 2)
{
  Impossible de charger le fichier ou l'assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' ou une de ses dépendances. Le fichier spécifié est introuvable.
  FileNotFoundException
}

Object: GHC_FloatingExecutionControl (level 1)
{
  Une exception a été levée par la cible d'un appel.
  TargetInvocationException
}

Object: GHC_FloatingExecutionControl (level 2)
{
  Impossible de charger le fichier ou l'assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' ou une de ses dépendances. Le fichier spécifié est introuvable.
  FileNotFoundException
}

Object: GHC_Execute (level 1)
{
  Une exception a été levée par la cible d'un appel.
  TargetInvocationException
}

Object: GHC_Execute (level 2)
{
  Impossible de charger le fichier ou l'assembly 'netstandard, Version=2.0.0.0, Culture=neutral, PublicKeyToken=cc7b13ffcd2ddd51' ou une de ses dépendances. Le fichier spécifié est introuvable.
  FileNotFoundException
}

sebandraos, 2020-01-27:
Hi @julienenpc 
Are you sure your Windows version is correct? 1607 EOL'ed in April 2019. If so, I recommend updating your Windows installation and retrying. The update will likely include an updated .NET Framework version that might fix some of the issue.
Thanks

julienenpc, 2020-01-27:
Je suis toujours en version 1607 après la mise à jour, c'est peut être ça le problème ?
https://github.com/HALRobotics/Beta/issues/69#issuecomment-578723520

tristangobin, 2020-01-27:
Could you try to update and then try to install HAL again? Let us know if this works.

julienenpc, 2020-01-27:
I just saw the DSI, we updated the computer and we are still in version 1607, there is no new update to do.
We tried to install it without the antivirus, and still the same problem. They don't know where it comes from.

sebandraos, 2020-01-27:
Ok, try installing the latest versions of [.NET Framework](https://dotnet.microsoft.com/download/dotnet-framework) and reinstalling the HAL Robotics Framework

julienenpc, 2020-01-27:
It's working ! Thanks a lot !

keshang-ctrl, 2021-03-17:
Zip File: 
[https://drive.google.com/drive/folders/1Mg1i_7-EbX9RloY4PExrekLIEzydAp-4?usp=sharing]

OS:
Windows 10 Home v2004

Versions:
HAL.McNeel V5 1.3.32-beta
HAL.ABB 1.3.42 beta 
Rhino 7 

Errors:
HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'.

DominikReisach, 2022-01-17:
Errors:
Nothing happens after opening Rhino or Grasshopper. Also, the HAL components do not show up in Grasshopper.

OS:
Windows 10 Home 21H2

Versions:
HAL.McNeel.V5 1.4.22-beta
HAL.UniversalRobots 1.4.17-beta
Rhino 7: Version 7 SR13

Zip File is attached to this post:
[HAL.zip](https://github.com/HALRobotics/Beta/files/7881194/HAL.zip)

sebandraos, 2022-01-17:
@DominikReisach Is anything written in the Rhino command bar during loading Grasshopper? Can you also please confirm that `HAL.ghlink` is present in the `%appdata%\Grasshopper\Libraries` folder?

DominikReisach, 2022-01-17:
The file is in the folder. Found the error message in the command bar:
HAL: Methode nicht gefunden: "Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()"

I guess that means conflicting dependencies (probably Enscape) and therefore #22 ? :)

sebandraos, 2022-01-17:
@DominikReisach Correct. If you have Enscape installed then that will definitely be the problem. Unfortunately that is an insurmountable problem at the moment so you'll need to deactivate Enscape whilst using HAL.

DominikReisach, 2022-01-17:
Thanks. It is working now!

XtreeE-PRO, 2022-02-23:
Hi,

Linked to the issue #120 you will find details below :

ZIP : https://we.tl/t-GlvaQnr3oT
OS : Windows 10 v21H1 Professional build 19043.1526
Rhino : v6 SR 35 (6.35.21222)
HAL : 1.4.22-beta

I think it appeared after installing this plugin : 

https://github.com/andrewheumann/jSwan

Thanks.

tristangobin, 2022-02-23:
This is probably linked to this issue https://github.com/HALRobotics/Beta/issues/22 since it seems using a different ([Newtonsoft.Json](https://www.nuget.org/packages/Newtonsoft.Json)) version.

XtreeE-PRO, 2022-02-23:
Thanks it worked with the "COFF" trick.

tristangobin, 2022-02-23:
As it seems resolved I closed this issue. Thank you for using our software and reporting issues.

closed 2022-02-23 by tristangobin

akaFlux, 2022-04-01:
Hi HAL not loading at all, had no issues before.
[HAL.zip](https://github.com/HALRobotics/Beta/files/8397941/HAL.zip).

akaFlux, 2022-04-02:
> Hi @akaFlux, your installation log does not show any error, do you have an error messages showing up in rhino when starting Grasshopper? Did you run the setup with your user windows account or with another acount? The ghlink was added there: C:\Users\josel\AppData\Roaming\Grasshopper\Libraries\HAL.ghlink

Hi @thibaultschwartz I have no errors messages, I do not see any HAL texts in the GH loading bar in the spalsh screen (I used to), I have ghlink in the libraries.

thibaultschwartz, 2022-04-02:
Hi @akaFlux, it really feels like you are not using the same account than the one you used to install HAL, or that you manually installed grasshopper somewhere else than its default location. 
- Never copy HAL files manually anywhere, the setup is much more complex. That's why there is a dedicated installer. The only thing that can be copied is the ghlink, but it should already be in you libraries, as I can see in your log it was created.
- You say "I see a text in the loading bar": then please copy it here. In rhino you can make the command bar as high as you want and you can open it as a separate window using F2.
- Are you sure you are using the same user account to launch rhino? For exemple: are forcing the execution as administrator, or did you execute the HAL installer as administrator instead of your regular user privileges?

AngelinaGaripova, 2022-06-22:
Hey!
I have an issue - HAL plugin disappeared from GH after Rhino 7 Update 
The error is:

The system cannot find the path specified. (Exception from HRESULT: 0x80070003)
HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'.

I have reinstalled everything from scratch several times and blocked Enscape plugin, but it is still not working.

ZIP file https://www.dropbox.com/s/n9326vgfaae387b/HAL.zip?dl=0
Rhino version 7 SR19 (7.19.22165.13001, 2022-06-14)
Windows Home 21H2 OS build 19044.1766

During the first installation, there was also a similar error, after which I blocked Enscape and everything started to work. But now after Rhino update everything stops working again.

Thank you.

sebandraos, 2022-06-22:
Hi @AngelinaGaripova , 
Your installation looks fine. I have just updated Rhino to 7.19 on my end and everything seems to be loading ok so it seems likely that a plugin other than Enscape is conflicting with the Framework.
Do you have any other plugins installed in Rhino and/or Grasshopper?

AngelinaGaripova, 2022-06-23:
Hi @sebandraos 
Yes I have other plugins installed, below are the pictures of them 
[image]
[image]

[image]
[image]

AngelinaGaripova, 2022-06-23:
Hi @sebandraos <https://github.com/sebandraos>
Yes I have other plugins installed, below are the pictures of them

thibaultschwartz, 2022-06-23:
Alternatively, run a search in your GH Library folder and look for "prism" or "unity"

AngelinaGaripova, 2022-06-23:
Thank you @thibaultschwartz !
I have tried to disable the solver for all the plugins and remove Heron and LunchBox from PackageManager - did not help.
Also, a search by "prism" or "unity" in GH Library gave nothing.
In the end, I uninstalled all Unity and Enscape apps from my computer and re-install Rhino and HAL from scratch and it helped. 
I don't know what helped exactly, but everything works and the error is gone.

xPhilippeRoux, 2022-07-06:
> Hi,
> 
> Linked to the issue #120 you will find details below :
> 
> ZIP : https://we.tl/t-GlvaQnr3oT OS : Windows 10 v21H1 Professional build 19043.1526 Rhino : v6 SR 35 (6.35.21222) HAL : 1.4.22-beta
> 
> I think it appeared after installing this plugin :
> 
> https://github.com/andrewheumann/jSwan
> 
> Thanks.

Hi,

I'm having a new error, still linked to jSwan. However the "COFF" is still enabled in the Developer Settings.

The error message is : 

> An error occured during GHA assembly loading:
>   Path: C:\Users\phili\AppData\Roaming\Grasshopper\Libraries\jswan_0.gha
>   Exception System.IO.FileNotFoundException: 
>   Message: Impossible de charger le fichier ou l'assembly 'Newtonsoft.Json, Version=12.0.0.0, Culture=neutral, PublicKeyToken=30ad4fe6b2a6aeed' ou une de ses dépendances. Le fichier spécifié est introuvable.

Details below : 

ZIP : https://we.tl/t-kHcCszMGCa
OS : Windows 10 v21H1 Professional build 19043.1766
Rhino : v7 SR 19 (7.19.22165.13001)
HAL : 1.4.41-beta

Thanks.

xPhilippeRoux, 2022-07-07:
Thanks @thibaultschwartz.

I went on https://github.com/andrewheumann/jSwan/issues/9, downloaded Newtonsoft.Json.dll & placed it in the Grasshopper folder. The error is now gone & jSwan plugin works fine.

However nothing on HAL Framework side. I uninstalled & installed again, same result.

ZIP file updated with current situation : https://we.tl/t-3PP7kZFQO3

xPhilippeRoux, 2022-07-12:
> Thanks @thibaultschwartz.
> 
> I went on [andrewheumann/jSwan#9](https://github.com/andrewheumann/jSwan/issues/9), downloaded Newtonsoft.Json.dll & placed it in the Grasshopper folder. The error is now gone & jSwan plugin works fine.
> 
> However nothing on HAL Framework side. I uninstalled & installed again, same result.
> 
> ZIP file updated with current situation : https://we.tl/t-3PP7kZFQO3

Any idea on this ? HAL plugin seems not installed whereas it is. My previous ZIP stays up to date.

sebandraos, 2022-07-12:
Hi @xPhilippeRoux ,
You have exactly identified the problem but unfortunately it's not resolvable - see #22 for more details.

xPhilippeRoux, 2022-07-12:
Hi @sebandraos ,
Could you elaborate a little ? I know that both plugins worked fine on my machine (with the "enable COFF" technique), and still work fine on other's. 

Are you suggesting to remove jSwan so that HAL can run, and that it is the only solution ?

sebandraos, 2022-07-12:
@xPhilippeRoux , this is a classic issue with plugin-type software. We and jSwan have dependencies on the same library - Newtonsoft.Json - but rely on different versions. Which one will work is purely dependent on the loading order of the plugins, the first one to load will load its version of Newtonsoft.Json and that will break the other. If others have both plugins working simultaneously then it's worth looking at what their COFF settings are and what other plugins they have which might be affecting the load order but it would be pure luck if it works.

One option which _might_ work is to trick jSwan by replacing their dependency with our version of Newtonsoft.Json from `C:\ProgramData\HAL` or vice versa. That is definitely not a supported scenario for the HAL Robotics Framework or jSwan but it _might_ work in your case.

krittikaw8, 2023-06-23:
[HAL.zip](https://github.com/HALRobotics/Beta/files/11849771/HAL.zip)
[image]
[image]
![20230623_HA
[image]
L configuration](https://github.com/HALRobotics/Beta/assets/57539283/c8c498dd-0b42-46bc-a162-65204ef6b01e)

Hello, 
Corresponding to the above mentioned error about which I emailed you today, I am attaching all the required information. 
I was obliged to remove the 'Extensions' folder from the zip folder, as it was too heavy to be attached here. 
Thankyou very much

sebandraos, 2023-06-23:
@krittikaw8 I can see that MathNet is definitely installed so it looks like this is a plug-in conflict as per #22 . 
Has our plug-in ever worked? If so, did you install anything recently that might align with when it stopped working?
If not, which other Rhino and Grasshopper plugins have you got installed? 
Try deactivating all of them and see if everything works then. If not, then we'll need to dig deeper. 

This is how you can disable Grasshopper plugins:

> HI Angelina, There is no way to identify the clashing plugin from theses screenshots. I would suggest you disable all other plugins using the GH loading settings to test if it's coming from a GH plugin at all: [image]

And you can do the same for Rhino from the Rhino Options -> Plugins Window. You can filter installed plugins by "Plugins that don't ship with Rhino" to find likely culprits.

[image]

You'll need to restart Rhino for that to take effect.

krittikaw8, 2023-06-26:
Hello, 

Thankyou very much for your answer.
It indeed helped fixing the issue. 

I removed MathNet and finally it works. 
HAL used to work perfectly fine before I recently installed some more plug-ins, so I guess that's where the problem occured. 

Thankyou very very much for helping me out.