HALRobotics/Beta #70: Crash when restart

6axis, 2020-02-18:
### Description

Rhino & GH crash if a restart order is given to the controller or if the robot is shutted down manualy (tested on ABB robot)

### Steps to Reproduce

1. Use the new controller cluster in GH to find your controller with the good IP Adress with the search function.
2. Shut down the controller using the breaker or warm restart it with robot studio 
3. Rhino crash

**Expected behavior:** Controller cluster go red or tell me their is an error. 

**Actual behavior:** Rhino crash

**Reproduces how often:** Look like every time

### Versions

HAL.Macneel.V5 - 1.2.48-beta
HAL.ABB - 1.2.51-beta
W10pro 1903

sebandraos, 2020-02-18:
@6axis Thanks for the report. I can reproduce this locally and have located the error. I am awaiting a review by @thibaultschwartz for a definitive fix and will then update the ABB extension and close this issue.

sebandraos, 2020-02-18:
@6axis Update on its way out now. Should be with you within the next half an hour or so. Please reopen this if the issue persists.
Thanks again for helping us make a better software.

closed 2020-02-18 by sebandraos