HALRobotics/Beta #71: Components visualization

uhart, 2020-02-27:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

Hello, I just installed the last version of HAL and I think that there are problems with visualization...
In fact, the robot and the tool components appear in Rhino only when the simulation is OK = solver completed. And then, when the simulation is executed, the robot does not move, only the TCP.

### Steps to Reproduce

1. Launch the KUKADemo.gh
2. Execute the simulation

**Expected behavior:** Previsualization of Robot and Tool components and movement of the robot during the simulation.

**Actual behavior:** No previsualization and the robot is static during the simulation.

**Reproduces how often:** 100%

### Versions

- **HAL.McNeel.V5** 1.2.52-beta
- **HAL.KUKA.ExtensionGroup** 1.2.43-beta

sebandraos, 2020-02-27:
@uhart Thank you for reporting this issue. From what I can see it appears to be a bug with the KUKA KR10 R900 sixx preset. I can reproduce the bug with the KUKA demo file but everything is working correctly in the ABB, UR and Staubli files, or if I change the robot preset in the KUKA file. We will look at the specific cause for this. Is this, unfortunately, exactly the robot that you have? If not, using your robot should fix the problem. 
If you manage to reproduce the error with any other robots please let us know.
Best,
Sebastian

sebandraos, 2020-02-27:
@uhart This is now fixed in the latest update. Please reopen if the issue repeats itself. And thank you once again for reporting these issues and helping us make a better, more stable software.

closed 2020-02-27 by sebandraos