HALRobotics/Beta #73: Robot script inside Grasshopper

anjakunic, 2020-06-04:
### Prerequisites

* [x] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

[Dear HAL team,

I would like to extract a resulting robot code into a panel inside Grasshopper. I remember this was possible in the old HAL version, but I don't seem to find a way to make it in the HAL robotics framework. Do you have any plans on integrating back this option?]
### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

**Reproduces how often:** [What percentage of the time does it reproduce?]

### Versions

You can get the exact version of the Framework and all installed extensions in the HAL -> About menu or via the installer. Please also include the OS and what version of the OS you're running.

### Additional Information

Any additional information, configuration, files or data that might be necessary to reproduce the issue.

thibaultschwartz, 2020-06-30:
Hi,
You can use the "Disassemble" component to retrieve the parts and joints of a mechanism, in case you need to reference one in particular. What data are you looking to extract and in what purpose?

thibaultschwartz, 2020-06-30:
Hi,
Apologies, I misunderstood your question in my previous reply. 
We added an output to the Export component recently which gives the path of all generated robot files, it allows users to use the Grasshopper "Read File" component to open the generated code file(s) and display them in GH like in HAL 5. It might not be public yet, in that case it will be in the next update we will be releasing over the next few weeks.
In the meantime you can use the "Export" component and browse to the folder you specified as a destination. Once you found your file, open it in notepad++, it will update when you export it again.

sebandraos, 2020-08-10:
This should be fixed in the latest update but please reopen if need be.

closed 2020-08-10 by sebandraos