HALRobotics/Beta #74: Unrecognized objects

Ngm0505, 2020-06-30:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

[Greetings,
After building my model into the HAL framework, I get "Unrecognized obje
[image]
[image]

cts" error message when I try to open my file again. If I try to dowload the missing plugins, I get "Couldn't find HAL for Grasshopper (1.2.59.0)] and "Not all plug-ins were found on the package server".
You can find attached an screeshot of the errors.
Best,
]
[image]
[image]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

**Reproduces how often:** [What percentage of the time does it reproduce?]

### Versions

You can get the exact version of the Framework and all installed extensions in the HAL -> About menu or via the installer. Please also include the OS and what version of the OS you're running.

### Additional Information

Any additional information, configuration, files or data that might be necessary to reproduce the issue.

thibaultschwartz, 2020-06-30:
Hi,
HAL must always be installed using the HAL installer, the Grasshopper package manager will **NOT** be able to install it.
I would suggest you uninstall and reinstall HAL.  Please make sure Rhino is closed when you do so, and that you have a stable internet connection.
Best,
T

Ngm0505, 2020-07-01:
Hi Thibault,
Works better. 
Thank you,

closed 2020-07-06 by thibaultschwartz