HALRobotics/Beta #76: HAL don't load when GH file opened form W10 explorer

6axis, 2020-09-20:
### Description

Rhino 6 bring the possibility to directly lauch GH file by double-clicking on it. When this time saving triks is used, HAL is not loaded properly.

### Steps to Reproduce

1. Launch a GH file with HAL components in it directly from the windows explorer

**Expected behavior:** HAL launch properly.

**Actual behavior:** GH tells that HAL components are missing.

**Reproduces how often:** 100%

### Versions

1.3.4 beta

tristangobin, 2020-09-20:
Hi 6axis, 

This is due to dependencies loading order. Unfortunately, we are aware of this bug but we can't fix it since it is related to a specific grasshopper behaviour.

We will keep you informed of any evolution.

Best,

sebandraos, 2020-10-08:
@6axis 
I'm currently unable to reproduce this issue on my desktop running the latest public version but don't have access to a laptop to ensure that this isn't just a timing issue with different speed computers.
Could you try updating your Framework installation by rerunning the installer and pressing "apply" and then trying this again? If the issue persists could you please let us know what the specs of the computer you're running are? 
Thanks

sebandraos, 2020-10-15:
@6axis Please reopen if this issue persists after the update

closed 2020-10-15 by sebandraos