HALRobotics/Beta #77: can not find method error (need help)

dngud, 2020-10-14:
Does any of you guys recognize this error?
'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'

after install it and I start my grasshopper this keep pop up and can not use HAL in GH.

sebandraos, 2020-10-14:
Hi @dngud ,
This sounds like a corrupt installation or an incompatibility with another plugin. Could you please follow the instructions in #69 to get us the data we need to examine why this is occurring? That same issue lists a potential workaround which will cleanly install the Framework. If that doesn't help, I suggest looking at #22 for possible plugin clashes.
Best,
Sebastian

dngud, 2020-10-14:
Thanks for fast reply! but those doesn't work for mine.....still every time I start GH that same message pop up and I do not see any of HAL functions.

sebandraos, 2020-10-15:
@dngud Without the information requested in #69 under **Help To Fix**, there's not a lot we can do to work out why this is occurring on your machine.

sebandraos, 2020-10-16:
I can't seem to reproduce this issue and need a bit more information, which is expected to be in every issue report, hence the template when you create a new issue. In particular here, I will need the OS version info, including the actual version number e.g. Windows 10 1809 (which on WIn10 you can find under Settings -> System -> About).

sebandraos, 2020-10-16:
Please also let us know which version of Rhino you're using.

sebandraos, 2020-10-28:
Closing as stale. Please reopen if this problem persists

keshang-ctrl, 2021-03-17:
Zip File:
[https://drive.google.com/drive/folders/1Mg1i_7-EbX9RloY4PExrekLIEzydAp-4?usp=sharing]

OS:
Windows 10 Home v2004

Versions:
HAL.McNeel V5 1.3.32-beta
HAL.ABB 1.3.42 beta
Rhino 7

Errors:
HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'.

sebandraos, 2021-03-18:
@keshang-ctrl Thanks for providing the requested files and sorry you're having issues.
This loads absolutely fine on our dev PC so this is likely to either be a clash between something you have installed and the HAL Robotics Framework, or an assumption we made about the installation on all client PCs that doesn't hold true. A couple of things that might get us closer to a solution:
1. Can you please try installing the [.NET Framework 4.8](https://dotnet.microsoft.com/download/dotnet-framework/net48)? It may already be installed, I can't remember if that's the case on Win10 2004, so if it tells you it can't install because it's already there then that's fine.
2. Can you send me a screenshot or some other list of the other Grasshopper and Rhino plugins you have installed? Ideally this would include a few items:
    - A screenshot of the contents of your Grasshopper Libraries folder (Grasshopper->File->Special Folders->Components) showing all the .dll and .gha files there.
    - A list of all installed Grasshopper plugins (Grasshopper->Help->About->Little black box in the top left corner of the popup)
[image]
    - A list/screenshot of all Rhino plugins (Rhino->Tools->Options->Plug-ins->In the dropdown menu select "Plugins that do not ship with Rhino")

With these we should be able to get a better handle on what might be causing this issue for you and hopefully get you back up and running asap.

keshang-ctrl, 2021-03-18:
@sebandraos I had tried to install the .NET Framework before because I saw it on another thread about this same issue but I do already have that installed. 
Here are the screenshots you asked for:
[image]
[image]
[image]

sebandraos, 2021-03-18:
Thanks @keshang-ctrl , that's spot on and shows that this issue is actually a manifestation of issue #22 . In this case it is the files you see here called `Prism.Wpf.dll`, `Prism.Unity.Wpf.dll` and possibly `System.Resources.Extensions.dll` that are causing issues. These appear to all come with `ghODS-v0.2.2.gha` as I can't find where that comes from and have looked at your other plugins which don't include those `.dll` files. Please try temporarily removing those files, incl. the `.gha` from that folder and let's see if that solves the issue. If it does, there is some advice in the #22 issue that might help to find a more permanent fix but ultimately this means the two plugins could be incompatible because they rely on different versions of the same dependencies. Clearly our plugin won't work with the versions in your Components/Libraries folder but you might get lucky and `ghODS...gha` might work with our versions. You can test this by removing only the `.dll` files and checking whether both plugins work as expected.

keshang-ctrl, 2021-03-18:
@sebandraos I removed the files and even did a fresh re-install of HAL but I still get the same error in rhino when i start grasshopper.
HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'.

sebandraos, 2021-03-18:
@keshang-ctrl That's odd. Let's try and narrow down which plugin we've got compatibility issues with. Could you try disabling all your Rhino plugins and emptying your Grasshopper Libraries folder (except for the HAL file) and then restarting Rhino to see if that solves the issue? If it does then replace all the necessary files in the Grasshopper Libraries folder and try again. If that also works then it's likely to be one of your Rhino plugins causing the issue. Try reenabling them one at a time until we find which one causes the issue. Thanks

keshang-ctrl, 2021-03-18:
@sebandraos That worked! It turns out its my enscape plugin for rhino. Is there a way to work around this or will I have to disable enscape while I use HAL?

sebandraos, 2021-03-19:
@keshang-ctrl For the time being I think that's the easiest solution. I tried installing Enscape this morning and immediately had the same issue so that definitely seems to be consistent. I have also had a bit of a dig around the Enscape dependencies and it seems like there's no guarantee we'll be compatible. We might get lucky and occasionally use the exact same version of Unity in which case we'll be fine but otherwise this is, as is stated in #22 , a problem all plugins potentially face and, unfortunately, there isn't very much we can do about it.

closed 2021-03-19 by sebandraos

keshang-ctrl, 2021-03-19:
@sebandraos That's unfortunate but not a big deal. Thankyou for all of the help!