HALRobotics/Beta #78: Solver broken after error encountered

mishaelnuh, 2020-11-24:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

Solver does not seem to properly run after an error is encountered from the procedure input of the controller. Toggling `solve` input or reinitialising the component does not seem to solve it. Closing and opening the file fixes the issue.

### Steps to Reproduce

1. Create a simulation with 2 or more procedures combined (using the merge GH component). [image]
2. Toggle the simulation on.
3. Unflatten one of the procedure input for the merge component causing error. [image]
4. Undo error. [image]

**Expected behavior:** The solve component should continue to run after the error is fixed.

**Actual behavior:** Execute component outputs error: "No executable procedures were found.". Reinitialising or toggling the solver input does not solve the issue. Placing a new solve component fixes this.

**Reproduces how often:** 100% reproducibility.

### Versions
1.3.20-beta HAL framework
Windows 10 VM using Parallels 16 on MacOS 11.0.1

### Additional Information
GH file to reproduce bug: [combine bug.zip](https://github.com/HALRobotics/Beta/files/5590233/combine.bug.zip)

theonlytruth, 2020-11-25:
I see the error message "No executable procedures were found." also often. I think the reason were other errors but also solving those didn't help to get the execute to run again. For me it worked to copy everything in a new file or restart the software.

Please look for a solution because this is for me the only thing which feels very "beta".

tristangobin, 2020-11-25:
Thank you to report it. We are looking into it closely.

TEMPORARY WORK AROUND :
It is not necessary to restart grasshopper or rhino. It should be enough to close the file and reopen it. Be sure that all files are closed. Sometime grasshopper open twice the same file.

sebandraos, 2020-12-04:
@mishaelnuh Thanks for reporting this, it isn't something we had noticed internally. We are working on fix at the moment but I just wanted to mention a useful little trick that would simplify your file: Passing a Point into a floating Target parameter (not full component) will automatically give you a Target facing downwards.
@theonlytruth Our initial exploration of this bug looks to be very closely related to inadvertently creating multiple Controllers and passing them to the Solver component. If you come across this issue again and you don't think it follows this same pattern, please try and send us a file that reproduces the bug so that we can fix it in your case too.
Thanks again to both of you for helping making the HAL Robotics Framework better for everyone

thibaultschwartz, 2020-12-04:
This issue is fixed in the next update. Please reopen if necessary.

closed 2020-12-04 by thibaultschwartz