HALRobotics/Beta #79: Arc preview after G-code import

6axis, 2021-01-04:
Hi, 

This is more of a question or a need for help regarding the Gcode Interpreter. 

I'm looking for a massive targets reduction in my programs by using as much MoveC as i can in a medium scale, low tolerances, milling application. For that, I want to use the ability of the Gcode Interpreter to understand arcs. No problem at all on this part. 

When the solver is done computing, the "simulated trajectory" is taking shortcuts and don't look like it's following arcs at all. However, in the execution control panel, MoveC are here as expected. 

Is that a just preview feature of arcs to make the simulation more fluid or is the robot really following the outputed path as it is showed ? 
Is that a "zone" or "blend" misenderstaning ?

If you need to, you can reproduce what I'm talking about with the attached files. The .iso is the Gcode to import. 
Sorry for the messy GH canvas.

Regards.

[Files needed.zip](https://github.com/HALRobotics/Beta/files/5765400/Files.needed.zip)

sebandraos, 2021-01-04:
@6axis I'm seeing jagged arcs when loading your file:
[image]

Is this what you're referring to or are you actually getting straight lines between each of your targets that should have arcs? These segmented arcs are a simplified representation of arc motions in the display mode used by default in the Framework. Let me know which issue you're getting and we can dig further.

sebandraos, 2021-01-04:
Also, just whilst I've got your file open, a couple of pieces of advice:
1. Try using an even more simple mesh of your tool, it will help with performance of the simulation and navigation during execution. Something like this makes a big difference to the number of faces without affected the visible geometry too much:
[image]
2. Try and start your procedures with a motion in JointSpace that isn't to a 0,0,0,0,0,0, it will avoid that singularity right at the beginning. We actually serialise all our robots out of their 0 positions to avoid that. In this particular scenario you might try putting your robot in the pre-milling approach position in JointSpace to ensure that configuration

sebandraos, 2021-01-07:
@6axis I have just pushed a new update to the Framework that should fix the issue. I'll close this issue for now as I believe it's corrected in this latest update but please reopen if you have any further issues with it.

closed 2021-01-07 by sebandraos

6axis, 2021-01-07:
@sebandraos Sorry for the lack of reactivity. 
I made a simpler exemple : 

[image]

In blue the preview from the simulation (preview from the combined component) and in white the preview from a "move trought target" after the Parser. But I think you answered my question : the arcs are simplified for preview purposes. I get arc motion at the end so I think their is no problem, this is just a little bit wierd because it didn't look like like an arc anymore on the preview.
My gess is that maybe some wayframes are use instead of endframe to generate the preview ?

Thanks a lot for the advices and sorry about the join configuration, the GH definition was still under construction. 

the files for this exemple : (follow the orange panel)
[Files needed.zip](https://github.com/HALRobotics/Beta/files/5782500/Files.needed.zip)

sebandraos, 2021-01-08:
This is definitely the simplification of the arcs as part of the preview. It is particularly visible on those smaller arcs which are typical of GCode imports because there is no notion of _blending_ in GCode. It would probably be better if you can export your GCode without those arcs and then use [blends](https://docs.hal-robotics.com/Overview/Glossary.html#blend) to maintain the speed of the robot across that path.