HALRobotics/Beta #95: Robot catalog empty (or close to)

XtreeE-PRO, 2021-03-31:
Hi,

As soon as I start Grasshopper and place my "Robot" component, I find out that there is almost no robot available (only the ones I have used recently I think).

Note that my first action was to quit Rhino, open HAL installer, remove everything and reinstall. It didn't change my problem.

### EXTENSIONS
- **HAL.McNeel.V5** 1.3.32-beta
- **HAL.KUKA.ExtensionGroup** 1.3.31-beta
- **HAL.ABB.ExtensionGroup** 1.3.42-beta

Rhino 6.34.21034.7001
Windows 10 Professionnel version 1909 build 18363.1440

I'm sure the issue has been covered in the section somewhere but can't find the answer.

Thanks.

Philippe

tristangobin, 2021-03-31:
Hi, 

Could you check your internet connection? 
Except the one you already used, robot, controllers and other element from our catalogs are  fetched online.

XtreeE-PRO, 2021-03-31:
Thanks for your reply @tristangobin , I confirm I was well connected when running the tests. I remember something about cleaning the cache memory ?

tristangobin, 2021-03-31:
Do you have a firewall activated, if so could you try to desactivate it?

XtreeE-PRO, 2021-03-31:
It just worked by deactivating the firewall yes. Thanks.