HALRobotics/Beta #96: Variable speed

XtreeE-PRO, 2021-03-31:
Hi,

I would like to export a module that does not contain any pre-defined speed but a variable that I choose instead.

In RAPID language I have :
`MoveL [[111.11637,198.629959,496.876908],[0,0,1,0],[0,0,0,0],[9E9,9E9,9E9,9E9,9E9,9E9]], [250,572.957795,5000,1000], [False,1,0,1,0,1,0], NullTool \WObj:=Root;`
And I want :
`MoveL [[111.11637,198.629959,496.876908],[0,0,1,0],[0,0,0,0],[9E9,9E9,9E9,9E9,9E9,9E9]], mySpeed, [False,1,0,1,0,1,0], NullTool \WObj:=Root;`

See screenshots attached : 
[image]
[image]

Thanks.

Philippe

tristangobin, 2021-03-31:
You should choose _PreDeclaration_ instead of _Inline_ and use a defined speed.

[image]

[image]

XtreeE-PRO, 2021-03-31:
Thanks.

In that case am I able to choose which variables to predeclare ?

I don't need to assign a name to all the targets, and I don't want to declare a speeddata that is already declared in my robot task (I just want to use it in the move instructions).

tristangobin, 2021-03-31:
So, the actual behavior is either you predeclare everything or you have inline declaration.  You can't choose which variable should be predeclare. I will mark this as Enhancement. 

It means that you are using a Task with speeds that are already set, but not by HAL. This raise a question, if you are using speed that are not declare by HAL, how can we simulate?

It would be nice to have a case study to understand exactly what is needed here and the reason behind this request.

tristangobin, 2021-03-31:
[image]

So, I think that it involves several things that are beyond the use of a variable in inline declaration. 
We should be able to mark a setting like a speed as _System variable_, something that is imported from a remote controller. However, we would need to match the _Aliases_ and replace. 
We do not have the capacity in the public version to import data from a controller. And also, this is only feasible on controller that allows such feature, which is not the case for all. 

I will leave this open and we will have a discussion internally on how we can propose something. 

Let me know your thoughts. This discussion can be pursued in a more private thread if this project involves intellectual properties.

XtreeE-PRO, 2021-04-06:
Thanks for this explanation @tristangobin.

My speeds are actually calculated inside my Grasshopper file, but not via a HAL component.

I would be happy to discuss this point along with discussion #94 about multiple toolpathes export. This is the main global subject. Shall we plan this meeting with you and @sebandraos ?