HALRobotics/Beta #106: After installing HAL and opening my rhino file, when i lunch the grasshopper, an error text appear in the comment line " HAL: Method not found: 'Unity.IUnityContainer Prism.Unity.Ioc.UnityContainerExtension.get_Instance()'."´´ "

kayjee93, 2021-06-27:
### Prerequisites

* [ ] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description

[Description of the issue]

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

**Reproduces how often:** [What percentage of the time does it reproduce?]

### Versions

You can get the exact version of the Framework and all installed extensions in the HAL -> About menu or via the installer. Please also include the OS and what version of the OS you're running.

### Additional Information

Any additional information, configuration, files or data that might be necessary to reproduce the issue.

tristangobin, 2021-06-28:
Hi, 
Could you check this issue if you did not already ?
https://github.com/HALRobotics/Beta/issues/22

Could you provide the plugin list for rhino and grasshopper?

Could deactivate all your plugin and check if HAL can load. If it loads, then we will have to identify which plugin generates a conflict. 

Kind regards,

sebandraos, 2021-07-01:
Duplicate of #77 and caused by #22. Please reopen with a full report if those issues do not match.