HALRobotics/Beta #107: Critical exception when configuring an ABB controller in Grasshoppe

POOLP3DP, 2021-06-29:
Hi everyone, when I try to use an ABB controller it gives me an error message.

tristangobin, 2021-06-29:
It seems to be related to this issue :  https://github.com/HALRobotics/HAL.McNeel/issues/373

POOLP3DP, 2021-06-29:
Thank you for your feedback Tristan. Unfortunately your link does not work

POOLP3DP, 2021-07-02:
Thank you for your reply. I have tried doing as you said and I still have the same problem. I tried to install previous versions of PCSDK but I still have the same problem.
[image]

POOLP3DP, 2021-07-02:
[image]
With this configuration it would seem to work

sebandraos, 2021-07-07:
Yes, avoiding the PCSDK and using RWS instead is definitely a valid option (and I personally default to RWS). As you have RobotStudio installed, you could try uninstalling the RobotCommunication Runtime 6.05 that we provide. It's possible that there is a conflict between it and the latest versions of RobotStudio.

closed 2021-09-15 by POOLP3DP