HALRobotics/Beta #112: Export procedure component asks for "solution" parameter

dinoclau, 2021-07-19:
### Prerequisites

* [X] Put an X between the brackets on this line if you have done all of the following:
    * Reproduced the problem in a minimal file that you can share if needed.
    * Checked that your issue isn't already filed: https://github.com/HALRobotics/Beta/issues?utf8=%E2%9C%93&q=is%3Aissue

### Description
[image]

We are trying to make a Rapid code from Grasshopper for an ABB IRB 2600, and everything is fine until we try to export, because the export component asks for a "solution" input. What should we connect to this input so that it runs?

sebandraos, 2021-07-19:
Hi @dinoclau , 
Knowing what the little red bubble on Export says would be useful as that, normally gives some indication of what the problem is. At a quick glance, although I can't quite read the text exactly, it looks like the text in you _Destination_ input is not a valid path. There appears to be a symbol preceding the actual path.

dinoclau, 2021-07-19:
Hi @sebandraos 
That was the problem! It was not a valid path, now it works ok :)
Thank you!!

sebandraos, 2021-07-19:
Glad to hear it!

closed 2021-07-19 by sebandraos