HALRobotics/Beta #113: An installation error occured

6axis, 2021-09-13:
Hello, 

It seem that I'm unable to update/re-install/install HAL on my computer. The installation process go well until this message pop-up : 
[image]

I unistalled the framework, installed it with a freshly downloaded setup, installed only the application with no extentions, and with all the extensions... no luck for now. 

Do you have any advices to do a deeper uninstall of HAL to (maybe) solve the issue ?

I had no problem doing the last update one my other computer and never had problem before using HAL on the one that is problematic today.
Rhino is in 7.9.21222.1500 official licence
W10 is 21H1. 

Best regard

sebandraos, 2021-09-13:
Hi @6axis , If you could send us over the installation log (C:\ProgramData\HAL\Installation.log) then hopefully we can see what's causing the installation to fail. Thanks.

6axis, 2021-09-13:
Hi @sebandraos, here it is : 

[Installation.log](https://github.com/HALRobotics/Beta/files/7156582/Installation.log)

Thanks for your help !

sebandraos, 2021-09-13:
There are a lot of errors like - `L'accès au chemin d'accès 'C:\ProgramData\HAL\HAL.Communication.dll' est refusé.` This sort of file access issue normally means that an installation was once run As Admin and now is not. Did you manage to install the update without extensions?
One option is to attempt to install as admin again but given that that's unnecessary to in the HAL Robotics Framework, I would suggest completely uninstalling (including the installer itself), then deleting:
`C:\ProgramData\HAL` and `%localappdata%\HAL` before reinstalling again from scratch.
Let us know how you get on.

6axis, 2021-09-13:
Ok ! it worked !

Yes I tried to install it as an admin in a desperate attempt to make the error disapear... But before that I didn't manage to install only the application without extensions.

So I deleted the two folders you said and reinstalled it normaly and the problem is now gone. Maybe the log is more explicit than me, it's way over my understanding ans skills !

Thank you so much for your help ! Hope it can help somehow.

sebandraos, 2021-09-14:
Glad to hear it @6axis . Don't worry, that documentation is a little tricky to find at the moment and the logs are there for exactly this sort of thing.

closed 2021-09-14 by sebandraos