HALRobotics/Beta #114: Installation error

XtreeE-PRO, 2021-09-21:
Hi,

I tried several times to update the framework but each time I'm getting an error without details such as the one below.

[image]

My last action was to uninstall the framework first (which succeeded), however it does not prevent the error from happening. The difference is that I don't have anything installed now.

I'm using Rhino 6 SR 35.

Thanks.

Philippe

XtreeE-PRO, 2021-09-21:
Here it is :

[Installation.log](https://github.com/HALRobotics/Beta/files/7204308/Installation.log)

sebandraos, 2021-09-21:
Thanks. There appear to be 2 slightly different issues here:
1. An access issue which I have just fixed and which will be sent out in an update to the installer this evening (by the time the changes work their way through the system).
2. You couldn't seem to access [this](https://pkgs.dev.azure.com/hal-robotics/Distribution/_packaging/HAL_Distribution/nuget/v3/index.json) page which is hopefully a momentary issue.

Please try running the installer again tomorrow morning (or at your earliest convenience) and hopefully that will work. If that doesn't fix your issue, please send us the logs again so we can see what went wrong and after you've sent us those logs you can delete `%LocalAppData%\HAL` and `%ProgramData%\HAL` to completely clean the installation and run the installer again.

XtreeE-PRO, 2021-09-22:
Thanks @sebandraos it worked !

closed 2021-09-22 by XtreeE-PRO

sebandraos, 2021-09-22:
Excellent, glad to hear it @XtreeE-PRO . Please keep the bug reports coming so we can keep improving the Framework.